Metrological/buildroot #81: JPEG images showing in QtWebkit with RPI2?

10robinho, 2015-02-14:
Hello,

I've rebuilded whole buildroot for Raspberry PI 2 and everything looks good except on QtWebkit browser I don't see JPEG images. PNG and GIFs are loaded.

I've tried with both -qt-libjpeg and -system-libjpeg and there is no difference.  
I've also tried to change fb depth for raspberry, but it's not helping.  
Of course, there is /usr/lib/qt/plugins/imageformats/libqjpeg.so on system.

Is there any way to debug this, for start?

albertd, 2015-02-18:
What browser are you using? qtbrowser?

10robinho, 2015-02-19:
I've tried with qtbrowser and this simple code:

    import QtQuick 2.4
    import QtWebKit 3.0
    
    WebView {
         url: "http://www.cnn.com"
    }

Everything looks good except there are no JPEG images. I don't see any log at all.

Also, I've made simple qml code that shows JPEG image and it's working (it is not QtWebkit):

    import QtQuick 2.4
    
    Image {
        source: "image.jpg"
    }

I have no idea where to look. Maybe QtWebit compilation was somehow optimized for RPI1, like some acceleration for JPEG decoding that doesn't work on RPI2?

abiehaf, 2015-02-26:
any progress here?

philn, 2015-02-26:
it works fine for us. QtWebKit doesn't use the OMX IL image decoders on the RPi.

abiehaf, 2015-02-26:
I ran qtbrowser with --webkit=2, but it couldn't display jpeg file and 
there are no errors on console.

---
Regards,

Abi Hafshin Alfarouq
http://hafs.in/

10robinho, 2015-02-26:
Now, I've rebuilded newest versions and it's working on my RPI2.
Still, I'm not sure what fixed it exactly, or to say it better, what broke it then.

@abiehaf maybe you can do update and rebuild again?

For me, it's fine to close this issue.

abiehaf, 2015-02-26:
Well, I'll try it

---
Regards,

Abi Hafshin Alfarouq
http://hafs.in/

closed 2015-02-26 by philn