Metrological/buildroot #88: WebRTC

Avnerus, 2015-05-12:
Hi,
First of all thank you for your wonderful work on this project. Probably the best way to run WebGL on the Raspberry Pi.
I wanted to compile in WebRTC support and stumbled into some problems.
I saw the referenced commit on the fork for WebRTC was quite old and tried to compile it with the newest commit but still stumbled into a lot of compilation problems. Some of which I was able to fix but eventually gave up.
Is there any way I can add WebRTC support without losing the WebGL support?
Also I was wondering, what are the changes in your qtwebkit fork as compared to the latest official version? Does the official version not support OpenGL ES2? 
Thanks!

/Avner

albertd, 2015-05-12:
We're adding WebRTC as we speak to the Webkit Wayland browser, we will keep you updated on the progress.

Avnerus, 2015-05-12:
Ah nice!
Do you find WPE to be better than QtWebkit?
Would love to help testing it. I see the current wpe config has Media Source enabled. Is that enough for webcam support or you also need Media Stream?
And another question, can the WPE browser run directly on the framebuffer the same way as QtBrowser?
Thanks.

albertd, 2015-05-12:
WPE is getting in a better shape than QtWebKit, furthermore the WPE is a few days behind WebKit trunk. We've done a partial MSE implementation on WPE, main features that we will be adding shortly is EME, MSE (rest) and WebRTC, the team has an intermediate version running I will check later this week if we can push it to the buildroot, so you could potentially test on that as well. WPE only uses Wayland without any desktop, the "rpi2_wpe_minimal_defconfig" even runs fully as a ramdisk and auto boots the browser to an url set in the wpe.txt

Avnerus, 2015-05-12:
Sounds great..
And yeah, if possible I would definitely like to test it.
Thank you very much!

/Avner

philn, 2015-05-13:
Please don't try the qtwebkit webrtc branch, it's an old failed attempt :)
You can indeed expect WebRTC support in upstream WebKit and WPE, I think by the end of the year in upstream and likely before in WPE... 
Don't confuse Media Source (MSE) with mediastream. MSE is about adaptive media streaming and mediastream is about WebRTC and exposing audio/video input devices in the browser.

Avnerus, 2015-05-13:
I see,
Thanks for the info!
So currently this repository: https://github.com/WebKitForWayland/webkit also does not yet support WebRTC? And what are the key differences between Metrological and WebKitForWayland? 
I'm a bit confused as to which repository I should follow if I'm looking for basic WebRTC and WebGL support on the Raspberry Pi

philn, 2015-05-13:
The Metrological repo contains changes that are more specific to the RPi and specific features needed by Metrological. The WebKitForWayland repo is more generic and contains changes that will eventually all be upstreamed in WebKit's SVN.

Currently the Metrological repo doesn't have WebRTC support yet. It depends on patches for GStreamer and OpenWebRTC that are not upstream yet either. So I'd say it's still too early to start testing this, unless you want to start playing with GStreamer/OpenWebRTC git master and the needed patches.

Avnerus, 2015-05-13:
Thank you for clearing it up.
There's a good chance I'd like to get involved in the effort but it'll probably be in few months from now.
In the mean time I'll keep on tracking the repositories.
How can I tell the status of a feature such as MediaStream? If I just look at the repository is seems that all of the relevant source and make files are already there.

philn, 2015-05-13:
Watch this repo: https://github.com/WebRTCinWebKit/webkit
Hopefully "soon" the changes made there will end up in WebKit's SVN :)

Avnerus, 2015-05-13:
Ah nice,
OK. Thanks!