Metrological/buildroot #89: GPU Memory Leak with movie

michaelhanin, 2015-05-21:
Hi,

I have a problem of GPU memory leak when looping multiple movie.

After a specific time ( always the same but  depending how much memory GPU is allocated ) movie stop working. ( with default configuration it crash at 33minutes )

GST_DEBUG show this error at this time.

gsteglimagememory.c:578:gst_egl_image_memory_setup_buffer:[00m Failed to create EGLImage

HTML assets used are available : http://welink.be/rpi2test.zip
GSTDEBUG is available here : http://welink.be/gst.log

I have test with the qtbrowser and athol, it seems to react the same (bad) way ...

Thank you

philn, 2015-05-22:
Memory usage also increases in Chrome here. Taking JS heap snapshots periodically shows the issue. So, not sure this is really a WebKit bug. The player seems correctly destroyed and recreated for each video.

michaelhanin, 2015-06-07:
Hello, 

After some change ( exactly same script but without create again the video element, just change the src of it, i have working uptime > 6hours ).

But with the last commit this example don't work anymore, it seems that gstreamer don't want to play local video ( file:/// or /.. ) but work with "http://XXXX/video.mp4" url .... strange

I will test to rebuild some commit ago to see if i can get this working again.

thank you

michaelhanin, 2015-07-31:
I use for workaround the change of attr src of existing video element ... thx

closed 2015-07-31 by michaelhanin