Metrological/buildroot #91: QtWebengine status?

10robinho, 2015-05-29:
Hello,

I've seen that QtWebKit will be depricated with Qt 5.5 and replace is QtWebengine.
I see that you have done some work on QtWebengine last year but not much lately.
So, question is, do you plan to migrate and what are main points that are making migration a problem?
Have you tried QtWebengine with RPI2, is it even able to build and work with eglfs backend?

I was thinking on working on it because for me QtWebkit is crashing all the time and I can't find proper way to fix it, so I'm looking for some solid starting points.

albertd, 2015-05-30:
@10robinho did you have a look at Webkit Wayland? Which is now part of the buildroot defconfigs and is in sync with the WebKit trunk.

10robinho, 2015-06-01:
Sorry if this is stupid question but is that "Webkit Wayland" still QtWebkit with qt having wayland as backend or something else?

philn, 2015-06-01:
WebKitForWayland is a new port, it has nothing to do with Qt. It is hosted there:
https://github.com/WebKitForWayland/webkit

And Metrological has a fork of that repo with some more changes.

closed 2015-10-20 by albertd

RoelRoel, 2015-11-18:
This sucks bigtime.