Metrological/buildroot #92: [wpe + athol] Strange behaviour after athol reloads the page after wpewebprocess crashes

depili, 2015-05-30:
We have a simple slideshow display, that works decently when wpe is first started, but which will crash after about 6 hours due to unknown memory leak.

The strange thing is that after the browser crashes and athol reloads the page (and we see a line: "[AtholShell] Crashed on 'http://example.com/iskdpy/', reloading." the webgl canvas only displays one pixel stretched to fill the entire screen. Apart of that canvas/webgl issue everything else seems to work as it should; the display communicates via websockets, runs the slide transition shaders etc but for some unknown reason only one pixel gets rendered.

I tried to look into athol sources to maybe find out what might be different after the page reload, but couldn't find the code responsible for the crash detection and reloading. We can reliably replicate this issue with wpe compiled at 22th of may, but each run takes 6 hours before the issue manifests.

Any pointers on where to look for possible differences between the first run and the reload after the crash?

albertd, 2015-05-30:
Could you share the example code you are running so we can do some debugging?

depili, 2015-05-30:
Will need to setup a new environment to get a "live" sample, as the display needs a back-end server connection via websockets. The code itself lives at https://github.com/iskcrew/iskdpy I will also test with a fresh wpe build tomorrow in case something has changed.

depili, 2015-06-06:
Seems like this has been fixed with the current wpe version, as a version compiled on 5th of July doesn't experience this issue anymore.

closed 2015-06-06 by depili