Metrological/buildroot #93: Strange issue with audio through RPI 3.5mm

flyandi, 2015-05-30:
Howdy,
Did anyone experience the same strange issue. I added the asta-tools to my buildroot and use amixer to set the output of audio to the 3.5mm (via amixer cset numid=3 1) and the speaker-test works as expected. However when starting either wpe or qt5browser and using web audio the sound always defaults to HDMI even when forced through the 3.5mm.

Anyone know how to solve this?

abiehaf, 2015-05-30:
I solved this by editing /etc/xdg/gstomx.conf
Try to change the value of rank in omxanalogaudiosink and omxhdmaudiosink

Regards,

Abi Hafshin Alfarouq
http://hafs.in/

flyandi, 2015-05-30:
Thx  - that worked!

closed 2015-05-30 by albertd

abiehaf, 2015-05-31:
You are welcome

Regards,

Abi Hafshin Alfarouq
http://hafs.in/

michaelhanin, 2015-06-27:
Hello,
which value need to be change for rank in /etc/xdg/gstomx.conf to set sound go to jack ?

thank you