Metrological/buildroot #94: WPE regression, missing jpeg support

depili, 2015-05-31:
A rpi2_wpe_defconfig compile done on 30th of May produced a wpe that can't display jpeg images. A compile done on 22th of May has jpeg support.

Tested with few random jpeg images, they all produce just a white screen, png images load fine and the same jpeg images load fine on the older wpe compile.

albertd, 2015-05-31:
Did you do a full clean build removing the output/ folder from Buildroot?

depili, 2015-05-31:
I did the usual "make clean && git pull && make rpi2_wpe_defconfig && make"

albertd, 2015-05-31:
Do you have an example page that has this issue? http://youtube.com/tv loads the jpeg's fine on my side.

depili, 2015-05-31:
For example this jpeg doesn't load for me in wpe: http://i.imgur.com/9IDHxyl.jpg

depili, 2015-05-31:
I will try a recompile with make distclean now also, but that won't be finished before tomorrow.

depili, 2015-06-01:
Did a recompile with e5ad389ddc4b4082eff7d476265af1acbc5b20e0

Ran make distclean before building, but still got a wpe that can't display (at least some) jpg:s, like http://depili.fi/tmp/1.jpg can't also load them into webgl textures and three.js gives the following warning:
"three.js:23400:19: CONSOLE WARN WebGL: INVALID_VALUE: texImage2D: bad image data"

zdobersek, 2015-06-04:
How are you trying to 'view' these images? Just loading them directly, or are you manipulating them through WebGL or something else?

depili, 2015-06-04:
Both trying to load it into a webgl texture or trying to just render the image with "wpe http://depili.fi/tmp/1.jpg" fail. Trying to load into a webgl texture results in the above texImage2D error and trying to just access the image with a straight url results in a white screen, accessing http://depili.fi/tmp/2.png by url or loading to a texture works and that very same jpg used to work just fine.

depili, 2015-06-04:
A compile of rpi2_wpe_minimal_defconfig done yesterday loads that jpeg again just fine. Previous errors where from the rpi2_wpe_defconfig, so it might be a difference in the configs, but after a quick look none of the differences should be jpeg or image library related. Both the previous broken environments and the current working environment have been compiled using the same virtual machine running debian jessie (aka stable).

closed 2015-10-20 by albertd