Metrological/buildroot #95: [wpe] Remote webinspector usage?

depili, 2015-06-01:
Commit 9070cad80e97b9e7c20ec0904920efd7636c3437 hints that buildroot should create wpe with webinspector support. But enabling the server in /usr/bin/wpe and connecting to it first gives a index of inspectable views (as it should) but clicking the link renders a blank page with tons of javascript errors in all browsers I have tried (firefox, safari, chromium).

Trying to connect to the web inspector also results in following errors in wpe console output:
"Error fetching webinspector resource files: 0, The resource at '/org/wpe/inspector/UserInterface/Protocol/InspectorBackendCommands.js' does not exist" so some resources seem to be missing from the wpe side?

philn, 2015-06-03:
It should work with Safari and Epiphany but the inspector is changing a lot in upstream currently, so don't expect it to be rock-solid stable :)

I've just tested it works here with a MiniBrowser build from WebKit trunk.

depili, 2015-06-03:
My try using safari on osx yosemite just produced a white page and when opening the javascript console in safari it was just full of js errors, will try the minibrowser later.

philn, 2015-06-03:
The inspector JS code is evolving rapidly towards ES6, your Safari version might not support it yet. Try with a Safari nighly snapshot maybe?

depili, 2015-06-03:
Tried with the nightly safari/webkit build, got way less errors, but still a "Failed to load resource: server responded with status 404 (not found)" in the nightlys log and the familiar "Error fetching webinspector resource files: 0, The resource at '/org/wpe/inspector/UserInterface/Protocol/InspectorBackendCommands.js' does not exist" in wpe console.

So for some reason my build is missing that InspectorBackendCommands.js

philn, 2015-06-03:
Yes it's weird... well I can't reproduce this issue here... @albertd does it work for you?

depili, 2015-06-03:
And for reference, I'm using a "rpi2_wpe_defconfig" compiled on 1st of June. Currently compiling a rpi2_wpe_minimal_defconfig to see if its a configuration issue somehow.

depili, 2015-06-04:
The new compile is still not working, I get the same error in the raspberry pi side:
"Error fetching webinspector resource files: 0, The resource at '/org/wpe/inspector/UserInterface/Protocol/InspectorBackendCommands.js' does not exist"

And webkit nightly on osx yosemite says:
[Error] Failed to load resource: the server responded with a status of 404 (Not Found) (InspectorBackendCommands.js, line 0)
[Error] ReferenceError: Can't find variable: RuntimeAgent
	RuntimeManager
	loaded
	(anonymous function)

depili, 2015-06-05:
I see a promising commit has appeared into WPE: https://github.com/Metrological/WebKitForWayland/commit/2bc07faff3c9d8b58e356246b8f73fae508badbb
Will recompile and test :)

depili, 2015-06-08:
The web inspector is now indeed working since that above commit.

closed 2015-06-08 by depili