Metrological/buildroot #96: wpe_minimal_defconfig and config.txt + cmdline.txt

depili, 2015-06-04:
Currently building the minimal wpe config results in a ramdisk, but the config.txt and cmdline.txt initialization files for rpi are just the same ones used for non-ramdisk configs. It would be more user-friendly to provide ready to go settings for booting to the ramdisk.

This would involve adding "initramfs rootfs.cpio.xz" to config.txt and removing the "root=" and "rootwait" parameters from cmdline.txt

I'm not familiar with the buildroot system so unfortunately can't provide a pull request :(

albertd, 2015-06-04:
Yes I agree, I will add it in the near future. I can share my configs here if you would like?

depili, 2015-06-04:
Not needed for me, with the above changes I have a booting wpe_minimal system. But for others and future here are my complete settings:

config.txt
```
disable_splash=1
disable_overscan=1
boot_delay=0
arm_freq=1000
gpu_freq=500
over_voltage=6
avoid_warnings=1
force_turbo=0
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=596
kernel=zImage
hdmi_group=1
hdmi_mode=4
initramfs rootfs.cpio.xz
```
cmdline.txt
```
dwc_otg.fiq_fix_enable=1 sdhci-bcm2708.sync_after_dma=0 dwc_otg.lpm_enable=0 vt.global_cursor_default=0 console=tty1 quiet
```

depili, 2015-06-04:
Actually digging the exact files out also brought up another possible enhancement, currently /dev/mmcblk0p1 isn't mounted to /boot on the ramdisk setup, so changing the autostart-url or hdmi mode requires manually mounting the boot partition first.

albertd, 2015-06-04:
Hereby my files:

cmdline.txt
```
dwc_otg.fiq_fix_enable=1 sdhci-bcm2708.sync_after_dma=0 dwc_otg.lpm_enable=0 vt.global_cursor_default=0 console=tty3 root=/dev/mmcblk0p2 rootwait quiet
```

config.txt
```
disable_splash=1
disable_overscan=1
boot_delay=0
arm_freq=1000
gpu_freq=500
over_voltage=6
avoid_warnings=1
force_turbo=0
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=396
kernel=wpe.img
hdmi_group=1
hdmi_mode=4
```

wpe.txt
```
http://youtube.com/tv
```

I renamed the zImage to wpe.img.

depili, 2015-06-04:
@albertd Your configuration isn't for booting the rpi2_wpe_minimal_defconfig that comes as a initramdisk, but for the rpi2_wpe_defconfig, since you mount a regular rootfs, and for that configuration the files from output/images/rpi-firmware/ are good. The issue is not having separate configuration files for the ramdisk defconfig.

albertd, 2015-06-04:
No, rpi2_wpe_minimal_defconfig also includes the ramdisk into the kernel, so I use just the zImage, not the rest please look at the file size.

```
dev@buildroot:~/buildroot$ ls -ltr output/images/
total 82104
drwxr-xr-x 2 dev dev     4096 Jun  3 17:03 rpi-firmware
-rw-rw-r-- 1 dev dev 43233792 Jun  4 04:43 rootfs.cpio
-rw-rw-r-- 1 dev dev 16613332 Jun  4 04:43 rootfs.cpio.xz
-rwxrwxr-x 1 dev dev 24216936 Jun  4 04:44 zImage
```

depili, 2015-06-04:
Oh, didn't notice that (and didn't actually know this was possible). So then the default config.txt etc should be mostly fine, just needs a little bit of documentation IMO.

closed 2015-10-20 by albertd