Metrological/buildroot #98: rpi2_wpe_defconfig and start URL

zehfernandes, 2015-07-07:
I compiled the buildroot with  `make rpi2_wpe_defconfig`  and Deployed in rpi SD card, but can't start any URL.

even wget works with internet sharing.

some config I missed?

DanielHudson, 2015-07-07:
Not strictly related to straight CSS but with `make rpi2_wpe_defconfig` I also get much much worse performance than shown in the following video on the same rpi2. https://www.youtube.com/watch?v=vpWYOu2NSF4

magomez, 2015-07-07:
I've tested that aprilzero.com web you mention and also other CSS animation pages and it they work fine in my RPi. And the same happens with the helloracer example.

It doesn't look like a software problem, but a hardware one. Curious. Can you check whether your power supply is enough? I'm using the RPi official power supply, which delivers 5V and 2A. Also, do you have many usb devices connected to the RPi that might be requiring too much power from it?

DanielHudson, 2015-07-07:
An xbox360 was acting as my power supply, I am going to order a good power supply and retry the default rpi2_wpe_defconfig again! I will post back.

flyandi, 2015-07-09:
I posted the video above and this was running the wpe_default_config. No issues really. I have a RPI2 with this config in production.

DanielHudson, 2015-07-09:
@flyandi out of interest was the hello racer loading screen flashing blue/black for you? (http://helloracer.com/racer-s/)

magomez, 2015-07-09:
@DanielHudson that happens because the loading screen uses the background as a progressbar of the loading process. The background is initially blue and black starts growing from left to right.

flyandi, 2015-07-09:
Yes, that's how it behaves.

DanielHudson, 2015-07-13:
So, I finally got around to rebuilding and testing this with a proper power supply - and it works much much better! So I'm assuming it was just a PSU issue for me :+1: 

Previously the loading bar was flickering blue/black, as opposed to working like an actual loading bar... But now it's all good, thanks all!

albertd, 2015-07-13:
Good news, closing ticket

closed 2015-07-13 by albertd