Metrological/buildroot #99: rpi2_qt5webkit_defconfig not booting.

DanielHudson, 2015-07-07:
I compiled the build root with `make rpi2_qt5webkit_defconfig` but it isn't booting, my Raspberry Pi2 just has a permanently lit green & red light when I try to boot it.

I have also compiled with `make rpi2_wpe_defconfig` which boots fine, but much like #98 performance is very slow.

magomez, 2015-07-07:
I've just double checked that the rpi2_qt5webkit_defconfig config boots perfectly. I was expecting that, as it uses the same kernel and drivers than the rpi2_wpe_defconfig config.

I would say that you are having a problem with the SD card detection. The situation you mention (steady red and green leds lit) is also what happens when you boot the RPi with no card inserted. I would try reformatting it and re-deploying the image and see whether it works.

In https://www.raspberrypi.org/forums/viewtopic.php?t=58151 you can find a list of problems when booting and how to solve most of them.

DanielHudson, 2015-07-07:
Thanks for checking, the same card appears to work fine with the rpi2_wpe_defconfig, and I've tried reformatting and re-deploying/making several times to the same effect.

But after what you say I am inclined to think it's just something weird with the SD Card, or Pi. I also need to try with a good PSU.

closed 2015-07-07 by DanielHudson