Metrological/buildroot #107: Blank page with remote inspector

madanagopalt, 2015-09-29:
Hi,

I am trying wpe wayland and opened remote inspector, only by the below command:

export WEBKIT_INSPECTOR_SERVER=<ipaddress:port>

I opened http://ipaddress:port in firefox. It is showing the opened URL in inspectable views. When i click that link,  JS Pages are downloaded but showing empty page. Please suggest a way to debug this.

Regards,
Madan

albertd, 2015-09-29:
Please use webkit nightly to view the webinspector

woutermeek, 2015-09-29:
Hi Madan,

You need to use the latest webkit nightly to view the inspector.
http://nightly.webkit.org/

Kind regards,

Wouter

madanagopalt, 2015-09-29:
Oh, ok, could you please tell, do i need any more settings to enable the webinspector?.

madanagopalt, 2015-09-29:
I am not seeing any script for run-wpe. I want to run it on weston compositor.Please suggest the way to do.

philn, 2015-09-29:
Setting WEBKIT_INSPECTOR_SERVER and using run-wpe in combination with the webkit nightly should be enough.

closed 2015-09-29 by philn

madanagopalt, 2015-09-29:
Hi,

I am working on linux. I am unable to see run-wpe on downloading the source code mentioned in site http://nightly.webkit.org/. 

Also, i tried executing the below commands:

1,
$ Tools/Scripts/update-webkitwpe-libs

The above script is not present.

2,
madan@madan:~/WebKit-r189384$ Tools/Scripts/build-webkit --wpe --threaded-compositor

====================================================================
 WebKit is now built (00m:00s). 
====================================================================

But i didn't get run-wpe binary. Please suggest a way to build. It has been mentioned, nightly buils is available only for Mac OS. If i want to build wpe-webkit for linux, please suggest the way to do. Thanks.

Regards,
Madan

philn, 2015-09-29:
You're not using the right git repository.

To use WPE on Linux you need to clone this repo: https://github.com/WebKitForWayland/webkit

philn, 2015-09-29:
To use the WebInspector on Linux your only option currently is a very recent version of webkitgtk (2.10 at least). The MiniBrowser should be able to act as a client to the WebInspector server.

madanagopalt, 2015-09-29:
Yes, i already tried with that repo. I got blank page, while running inspector on firefox. Is it like, today i have to download and build once?. Please clarify.

Regards,
Madan

philn, 2015-09-29:
Firefox isn't compatible with WebKit's web inspector.

madanagopalt, 2015-09-29:
Ok, could you please tell what browser i can use on linux to check the web inspector.

madanagopalt, 2015-09-29:
I tried with another wpe browser also and behaviour is same, could you please suggest what browser i can try ?

madanagopalt, 2015-09-29:
Please let me know,which browser I can use in Linux for accessing webinspector in wpewebkit.

philn, 2015-09-29:
Did you read https://github.com/Metrological/buildroot/issues/107#issuecomment-144053250 ?

madanagopalt, 2015-09-29:
Ok,so I have to check only with gtk and not with wpe?.Any release is planned for inspector working with wpe,because I am using wpe webjit and not gtk. Or any configurations I might have missed?.please suggest.

philn, 2015-09-29:
This is the only solution currently. Build webkitgtk trunk and use MiniBrowser: http://trac.webkit.org/wiki/BuildingGtk

madanagopalt, 2015-09-29:
Ok,thanks