Metrological/buildroot #108: rdate tries to update the clock before network is ready

depili, 2015-10-06:
Currently init first starts the network initialization process and then tries to update the RTC from time.nist.gov. This has a problem in that dhcp lookup doesn't block the init process and thus in most cases the time lookup fails as the network isn't up yet.

```
Starting logging: OK
Populating /dev using udev: done
Initializing random number generator... done.
Starting network...
rdate: bad address 'time.nist.gov'
udhcpc (v1.22.1) started
Sending discover...
Sending discover...
Sending select for 192.168.0.30...
Lease of 192.168.0.30 obtained, lease time 6000
deleting routers
adding dns X.X.X.X
adding dns X.X.X.X
Starting dropbear sshd: OK
Starting WPE: OK

albertd, 2015-10-06:
It will automatically retry when network is connected

depili, 2015-10-07:
Oh you are right, probably had a network glitch last night when testing as the clock didn't update to current time on few occasions, seems to work now.

closed 2015-10-07 by depili