Metrological/buildroot #111: rpi2_wpe_minimal_ml_defconfig build fails due to cppsdk repository missing

usultis, 2015-10-31:
I tried to build the rpi2_wpe_minimal_ml_defconfig build but it fails due to cppsdk repository missing. 
I also tried to build rpi2_wpe_bluetooth_defconfig, but it fails due to the same reason.
Building rpi2_qt5webkit_defconfig works ok 

Build log:
```
make[1]: Leaving directory `/mnt/data/jenkins/workspace/bdl-screen-os-ml/output/build/cairo-1.14.2'
[7m>>> cppsdk 01bb1175b7cba3363b1dd9ddde20e29e62f49b0d Downloading[27m
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Doing full clone
Cloning into bare repository 'cppsdk-01bb1175b7cba3363b1dd9ddde20e29e62f49b0d'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
--2015-10-31 12:39:04--  http://sources.buildroot.net/cppsdk-01bb1175b7cba3363b1dd9ddde20e29e62f49b0d.tar.gz
Resolving sources.buildroot.net (sources.buildroot.net)... 176.9.16.109
Connecting to sources.buildroot.net (sources.buildroot.net)|176.9.16.109|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-10-31 12:39:04 ERROR 404: Not Found.
```

anything I could do to workaround this issue?

bramoosterhuis, 2015-10-31:
Hi Ryszard,

Try disabling cppsdk in "make menuconfig" Location: Target packages ->
Libraries-> Other. You might have to disable "dxdrm" first (if set) under
digital rights management.

good luck!

Groetjes,

Bram

www.bramoosterhuis.com

Voeg mij toe aan jou netwerk:

[image: Facebook] <https://www.facebook.com/bram.oosterhuis> [image: Google
Plus] <https://plus.google.com/109279857157290355884/posts> [image: Twitter]
<http://twitter.com/#%21/bram1984> [image: LinkedIn]
<http://www.linkedin.com/profile/view?id=46639075&trk=tab_pro>

Contact gegevens:

albertd, 2015-10-31:
Please use the rpi2_wpe_minimal_defconfig without "ml". This is an internal defconfig

closed 2015-10-31 by albertd